Patribots4738/ChargedUp2023 #79: Clamp the odometer for the swerve so it is harder to overshoot

GalexY727, 2023-03-14:
if X < 2 (for example) and we are on blue alliance, set X = 2, and if Y is < 0, set it to 0; if Y > FIELD_HEIGHT_METERS, Y = FIELD_HEIGHT_METERS.

Note that the X value must only be clamped on the low end on blue alliance, and only clamped on the high end on red alliance.

PatribotsProgramming, 2023-03-17:
Merged into #96

PatribotsProgramming, 2023-03-17:
This resulted in some strange reaction to the robot as if resetting the odometry was taking a lot of time

PatribotsProgramming, 2023-03-17:
@Jacob1010-h :(

GalexY727, 2023-03-17:
1 am solutions: only update the odometery if the requested value is within the range
i.e
if (swerve.getPose() < LIMIT) {
  update odometery 😮 
}

GalexY727, 2023-03-17:
@Jacob1010-h :)

Jacob1010-h, 2023-03-18:
Needs Testing???

GalexY727, 2023-03-18:
thanks jacob 😭

GalexY727, 2023-04-09:
While this PR never really got closed, I believe 2903f19e26ce55b09dfec7f0c0d26ebca35133b2 resolved this in full.

closed 2023-04-09 by GalexY727