Patribots4738/ChargedUp2023 #113: Set max angular velocity based on our arm X position

GalexY727, 2023-03-27:
use FK to slow down spin when extended but speed up when stowed

PatribotsProgramming, 2023-03-28:
[desmos] (https://www.desmos.com/calculator/fqyyldertp) :>
Apply a deadband at 6.5, anything less than that is fully within the frame perimeter

PatribotsProgramming, 2023-03-28:
arm X of ~30 inches should return something around 3 PI max angular speeds, and an arm X of ~0 inches (remember the deadband will return zero if anything under 6.5 is given) our max angular speed should be roughly 6-8-10PI (needs testing)

PatribotsProgramming, 2023-03-28:
java "Lerp" should suffice
pseudocode:
`import java.util.Lerp;
maxAngularVelocity = Util.Lerp(MIN_ANGULAR_SPEED, MAX_ANGULAR_SPEED, Math.clamp(MathUtil.applyDeadband(arm.getXPosition(), 6.5)/30, 0, 1)`

GalexY727, 2023-04-03:
60f9a4dabe00e0a46b24fd49ea7905734988b168

closed 2023-04-03 by GalexY727