Patribots4738/ChargedUp2023 #117: Lower the cube intake to be closer to the ground

PatribotsProgramming, 2023-03-28:
Ideally, this will make it so that we are able to contact the cubes quicker and thus intake with more precision

GalexY727, 2023-04-03:
After a full day of testing, I think that the better solution is to index the claw against our bumper to prevent swinging, as seen in our 3P autos

GalexY727, 2023-04-03:
Meaning that this is ready to test :o

GalexY727, 2023-04-09:
I am content with its current position, but we can easily tweak it in pits/practice fields at houston

GalexY727, 2023-04-09:
gosh that was weird to type

GalexY727, 2023-04-09:
h

o

u

s

t

o

n

GalexY727, 2023-09-12:
More modifications were made today, after countless misses at houston I don't know how it never works at comps. It seemed okay enough today in testing so... marking this as complete until we go to chezy and realize it never worked in the first place...

closed 2023-09-12 by GalexY727