Patribots4738/ChargedUp2023 #128: Tune the FF gains for our shwerb

PatribotsProgramming, 2023-04-06:
Using sysID, find and tune FF gains for the motors controlling our swerve. [This CD post](https://www.chiefdelphi.com/t/whats-a-reasonable-target-for-swerve-module-response-time/397042/7) has a couple of suggestions. including adding ` Math.signum(driveOutput) * kS` to the kS gain.

PatribotsProgramming, 2023-04-06:
[This post](https://www.chiefdelphi.com/t/does-sysid-work-with-swerve/397605) says that is impossible, but we both know that is so false! Just use code to set all the wheels "up" and use sysID to make it better and better, so gooder :>

PatribotsProgramming, 2023-04-06:
[This post](https://www.chiefdelphi.com/t/applying-sysid-feedforward-to-an-sds-swerve-mk4/398455) has some nice images that should help us get through the hard parts of setting this up

PatribotsProgramming, 2023-04-06:
OK, [last one](https://www.chiefdelphi.com/t/sysid-feedforward-for-velocity-rpm-control/405516)
This one talks about FF and velocity, instead of the more popular voltage control type

GalexY727, 2023-04-07:
What if we used the [PPHDC ](https://cdn.discordapp.com/attachments/932511922052550667/1093792704720273428/image.png)@Jacob1010-h :o

GalexY727, 2023-04-09:
That PPHDC was really wild... I am going to save that for an offseason, or next year, or similar... but it has potential. Unfortunately our autos this year are very _egregious_, so we need to spin fast fast fast. This can do that, at the cost of us being less accurate in our position later down the line for picking up. I think that if our paths were more reasonable to see in the graphs of PP, this would be very beneficial for us.

But, since we are packing the robot as I type this message, we will need to postpone our usage of this new HDC

Also, I wanted to keep note that while this HDC was nice, it _required_ a state to run the calculate() method, which we use in teleop alignment, meaning that we would need both HDCs anyways even if we wanted to use the PPHDC for auto and HDC for teleop

closed 2023-04-09 by GalexY727