Patribots4738/ChargedUp2023 #131: Maybe lower telemetry data to utilize more of CAN bus

PatribotsProgramming, 2023-08-31:
https://docs.revrobotics.com/sparkmax/operating-modes/control-interfaces#periodic-status-frames 

I want to go over this and its importance, Reza knows more abt it than us.

GalexY727, 2023-09-12:
@Jacob1010-h I need to ping you because i keep forgetting about this

also can you remind me to look at the periodic status frame of the absolute encoders for the swerve modules and see if we can get that lowered for a faster updating odometery i.e. gooder autonomous

GalexY727, 2023-09-13:
https://docs.revrobotics.com/sparkmax/operating-modes/control-interfaces#periodic-status-5-default-rate-200ms
I would definitely make this faster than 5hz since 5hz is... well... 5hz. It may help with accuracy in autonomous. Thoughts @Jacob1010-h ?

Jacob1010-h, 2023-09-13:
seems like it could be a reasonable option, I'm just worried about any slots being taken over by re-allocating the default rates on the processor.

GalexY727, 2023-09-13:
If that matters we can just throw away non telemetry data and push it back

PatribotsProgramming, 2023-09-13:
Resolved in 9f97cf88a1f92c91a2a5a4c1c19cbba473d09232

closed 2023-09-13 by PatribotsProgramming

Jacob1010-h, 2023-09-14:
I love this !!