Patribots4738/ChargedUp2023 #135: Bezier Curves for Arm positions

Jacob1010-h, 2023-09-12:
**Describe the solution you'd like**
Instead of what we have now, which is creating an array of x,y positions, ~4 for each node; create an array of Translation 2ds, abt 0.02 ms appart in value t --> (0,1) that comprises the standard model for a quadratic bezier curve, shown in this desmos example. 

This is a cool idea, and I will begin implimentation soon after this commit. 

Co-Authored: @GalexY727

Jacob1010-h, 2023-09-12:
I think I finished this [branch](https://github.com/Patribots4738/ChargedUp2023/tree/135-bezier-curves-for-arm-positions) thank you @GalexY727  for the periodic help 🙏 🪐 🚀 ✨

Jacob1010-h, 2023-09-12:
https://www.desmos.com/calculator/j3shfjmzt3?lang=de

GalexY727, 2023-09-12:
this isn't closed silly we haven't even tested

Jacob1010-h, 2023-09-13:
I have an idea for a fix. Right now we see that the velocity of our arm is kind of weird, however, our FF has a calculated method that takes in an input of a velocity, it is possible, using the sample function as described in 048c013

PatribotsProgramming, 2023-09-15:
We tested yippee arm trajectory branch on top

closed 2023-09-15 by PatribotsProgramming