Patribots4738/ChargedUp2023 #137: Mid cone placement accuracy adjustment

GalexY727, 2023-09-12:
I wanted to shed some light on this since I noticed us missing two cones on our final finals match at Houston, both of which being on the mid row. I think that this is a product of two things:
1) the claw is too far and the cone is not centered on the node
2) upon stowing after placing, the claw moves straight up, as if it were trying to avoid the piece. This ends up giving an opportunity to let the cone slip back off since there may still be friction, instead we should immediately stow as denoted in the commit of adding Index 24 to the array of arm positions.

Although there is a commit on this matter, It may not be completely resolved, as there is still a possibility that the claw is now too close to the robot which would also result in a faulty placement.

GalexY727, 2023-09-12:
commit referenced above [^^^](https://github.com/Patribots4738/ChargedUp2023/commit/52cd0a4aea882fabff12f02538daf77fec6dcb80)

PatribotsProgramming, 2023-09-21:
Tested and approved

closed 2023-09-21 by PatribotsProgramming