Patribots4738/ChargedUp2023 #138: Test getTilt()

GalexY727, 2023-09-12:
I did some modifications to getTilt, and I want to see if there is still the sliver of a chance that it works since that would be a cool last minute addition. If it doesn't work, oh well.

GalexY727, 2023-09-12:
Expected outcome: the robot will always return a orientation relative to the charging station, no matter the holonomic rotation. This should use a really cool combo of quaternions and dot products but we will see if it wants to work on us. The math has been proven, if we are really desperate we can get Espinoza to help.
(in all reality it's only useful for auto but even then our other method works just fine)

That makes me wonder if we should look at our auto leveling code and see if it's still working fine... i'm sure it will be but... you never really know now do you

PatribotsProgramming, 2023-09-15:
Didnt work :(

closed 2023-09-15 by PatribotsProgramming