Patribots4738/ChargedUp2023 #146: Implement PS5 Controllers rather than Xbox

GalexY727, 2023-10-05:
With hopes that we have some future upgrades for our drivers...

GalexY727, 2023-10-05:
It will be required to use [ds4controller ](https://ds4-windows.com/download/ryochan7-ds4windows/) which will need to be installed on the driver station. Other than that, [this cd](https://www.chiefdelphi.com/t/ps5-dualsense-controller/401506/5) post explains that it works just the same as the ps4 controller class. Rubmles must be handled through the Xbox controller class, unfortunately.

GalexY727, 2023-10-05:
POV also works the same, so that makes [OICalc](https://github.com/Patribots4738/ChargedUp2023/blob/main/src/main/java/calc/OICalc.java) a lot easier to translate. We might not need to use [OICalc.toCircle()](https://www.desmos.com/calculator/e07raajzh5), though.

GalexY727, 2023-10-05:
@Jacob1010-h Do you have time to do this or should I? We have a branch [here](https://github.com/Patribots4738/ChargedUp2023/tree/146-PS5-Controller) which is ready to work on.

GalexY727, 2023-10-06:
Major progress with [964e07c](https://github.com/Patribots4738/ChargedUp2023/commit/964e07cdb640d9a4217ad574a4d990bb4f975e37)

GalexY727, 2023-10-07:
using DS4 there are no changes needed in code

closed 2023-10-07 by GalexY727

Jacob1010-h, 2023-10-07:
> @Jacob1010-h Do you have time to do this or should I? We have a branch [here](https://github.com/Patribots4738/ChargedUp2023/tree/146-PS5-Controller) which is ready to work on.

Sorry for not responding right away, i was distracted with family stuff, thank you for getting this done

GalexY727, 2023-10-08:
All good, don't worry about it