Patribots4738/ChargedUp2023 #150: Stow after 3p auto

GalexY727, 2023-10-16:
poor arm gets stuck on the field when we teleopinit on a 3p

GalexY727, 2023-10-16:
the reason it does this is because we are telling the arm to stow at telopinit but due to our arm being limp right after auto ends for 3 seconds the arm gets stuck on the grid and then just tries to go into the bot, not realizing there is a grid in the way
The solution to this is one of two things. Either
 
a) make a new desired arm position to be mid place instead of stowing or
b) stow in auto before the arm has time to go limp

It would likely be best if both could be implemented as we sometimes finish auto right on the buzzer and sometimes we have a second to spare. It all depends.

GalexY727, 2023-10-16:
If a new programmer would like to try to tackle this and learn how the code works I would love to help explain the implementation of this solution in further detail!

Oliver-Cushman, 2023-10-19:
If we wanted to stow the arm at any point would we use:
`arm.setArmIndex(PlacementConstants.STOWED_INDEX);`
?

PatribotsProgramming, 2023-10-20:
Yes, that would work, you would just need a way to see if all of auto is over, then run that :)

Oliver-Cushman, 2023-10-20:
Do we want to stow at the end of ONLY 3p autos?

GalexY727, 2023-10-22:
It's fine to do it after any auto

GalexY727, 2023-10-22:
I was thinking it would have something to do with  autoSegmentedWaypoints having no more waypoints

GalexY727, 2023-10-22:
which i think we currently use to run auto charge pad leveling code, just tack this onto the side maybe

GalexY727, 2023-10-24:
69a8b60f7c7849f3409723d5dbdc5ce8b3b8cd87

closed 2023-10-24 by GalexY727