Patribots4738/ChargedUp2023 #151: 3P auto with third one being low :D :dagger:

RudyG252, 2023-10-18:
Currently all our 3 piece autos consist of 2 highs and one mid, but in some matches, specifically in quals, our alliance members can't necessarily score mid/high consistently, or prefer to score low, so us scoring mid in auto instead of hybrid either forces us or our teammates to waste time scoring two pieces mid for a link instead of working on filling out high, or forces us to ignore that one piece mid which causes us to possibly lose a link in hybrid, just for 1 extra point in mid. For example, if we look at qual 47 at Batb, we lose a ranking point because we fill out the top row, have one piece in the middle, and have the entire bottom row filled besides one piece, which means if our mid piece was scored in hybrid we would have theoretically gotten that 6th link. This shouldn't have any drawbacks in matches where are strategy isn't focusing on mid, and could possible get as an extra link/ranking point in quals :D.

Qual 47: 
[image]

PatribotsProgramming, 2023-10-18:
Holy crap rudy i love you man

GalexY727, 2023-10-19:
This one is ready to test, should be able to test just on the robot cart once everything is put together and ready

GalexY727, 2023-10-19:
(I forgot to commit and make a branch and things so I do apologize for that...) Its saved on the programming computer though.

GalexY727, 2023-10-26:
This workie yey [0a2a7d7](https://github.com/Patribots4738/ChargedUp2023/commit/0a2a7d73d2d7f2a26d823b71fee72a1099eeb1d2)

closed 2023-10-26 by GalexY727