Patribots4738/ChargedUp2023 #158: Rewrite Driver UI and the other logging things

Jacob1010-h, 2023-10-29:
Some of the changes have been made from d9dc5dc35fac8a5b8245695213182fa1fdca371b however not everything!!

Jacob1010-h, 2023-10-29:
@GalexY727  Here is the current view of the data values, Do you want any more file management? 
[image]

Jacob1010-h, 2023-10-29:
everything has been moved out of DriverUI, however I really think that some type of file management might be needed
[image]

Jacob1010-h, 2023-10-29:
No file management:
[image]

File management:
[image]

Jacob1010-h, 2023-10-29:
It's up to you @GalexY727, but if you want file management, pull from: 51044c07d98c06f240f754490189d4582f081659
no file management: 8512f62c7711ddc681208c6f4e1c6ebc557b39b7

GalexY727, 2023-10-29:
[image]
I think that this would be a good idea

Jacob1010-h, 2023-10-29:
need to do alex folder thing

Jacob1010-h, 2023-10-31:
ngl I forgot that this was a thing for a bit, ig thats what happens when you do late night coding lol

this was resolved with this commit 41ea73411fe051ed0ee45accfb0ebe881464c03d and allows for Drive UI to still exist because of stupid ShuffleBoard